LaserWeb/deprecated-LaserWeb1 #101: Firmware undefined

Downunder35m, 2016-02-04:
I am running an Arduino Mega with the Tyrnkey Tyranny firmware.
Whenever I start the server it says "Found device: Arduino LLC (www.arduino.cc) Firmware Detected: undefined Port ID No: 0".
At first I thought this might be because I used a clone, so I swapped the board for a genuine Mega without the CH341 chip but the result stays the same.
I also checked the report string in language.h but again no firmware detected.
So I change the string back to the original: "FIRMWARE_NAME:Marlin " DETAILED_BUILD_VERSION " SOURCE_CODE_URL:" SOURCE_CODE_URL " PROTOCOL_VERSION:" PROTOCOL_VERSION "

At this stage I simply assume it is something in the firmware, so I would like to know if there is anything I need to change so the server can recognise my board correctly?
Today did a new pull from Git to make sure I am on the latest version of all.

openhardwarecoza, 2016-02-04:
Do a connect in Pronterface and copy paste me the entire startup dump please.

Downunder35m, 2016-02-04:
Wow, if there is one thing you are than it is fast to reply :)
Here you go:
start
Printer is now online.
echo: External Reset
Marlin 1.0.0
echo: Last Updated: Feb  4 201623/01/2016 08:50h EAST 14:41:23 | Author: (Downunder35m, Laser V1.1)
Compiled: Feb  4 2016
echo: Free Memory: 3587  PlannerBufferBytes: 2576
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X157.48 Y157.48 Z6047.24 E500.00
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z10.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X600 Y600 Z2 E2
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S600.00 T600.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Laser lifetime usage:
echo: Hours: 0
 Minutes: 0
echo:PID settings:
echo:   M301 P22.20 I1.08 D114.00

cojarbi, 2016-02-04:
Note that when I connect my ramps with that firmware I hade no connection issues. The only thing I had was laser on inmediatly since I did not mod my ramps

openhardwarecoza, 2016-02-04:
I dont see anything wrong on that output

You can force marlin mode by running 
node server.js --debug Marlin

Downunder35m, 2016-02-04:
Cheer, will try that in the afternoon when I am back on my computer :)

Downunder35m, 2016-02-05:
Update: Forcing Marlin for the server worked a treat, connection is now no problem anymore.
However I noticed that the Y-axis is still going up in Laserweb although my hardware is the other way around.
So all images are turned upside down and it quite hard to setup but at least I can test it now and compare some useful results.

openhardwarecoza, 2016-02-05:
We use CNC standard: lower left = 0,0  and positive moves are up/right.    Fix your motor wiring

closed 2016-02-05 by openhardwarecoza

Downunder35m, 2016-02-05:
No offence but if you have no clue you should not give ill advise.
The switched coordinate system is no matter of motor wiring but overall firmware configuration.
I do apreciate the help here but the first sentence of your answer would have been suffient ;)

openhardwarecoza, 2016-02-05:
However you like it. I would assume your firmware is correct (dir reversed == false) and you made a mistake wiring. Please feel free to stop asking for advice then (: