LaserWeb/deprecated-LaserWeb1 #103: Origin location

funinthefalls, 2016-02-07:
A large number of K40 users could benefit from LaserWeb, but as you are aware the K40's do not follow the CNC standard of 0,0 being in the bottom left corner, for the K40's it is the top left corner. Moving the end stop and changing motor direction may be a little too much for some users to do, so I was wondering if we could have a feature to toggle where 0,0 is placed. Not sure how difficult it would be for you to implement it as a software change.

openhardwarecoza, 2016-02-07:
To me that feels like more of a firmware use case, but more importantly, if you have gone through the effort to convert a machine (exactly to fix the chinese lack of standards) why not just fix the endstops too (swop min and max connectors) or motor wiring for Y axis?) while you are at it (:

funinthefalls, 2016-02-07:
K40 only has min endstops, no max endstops so the conversion requires making a new mount and physically rerouting the wiring (difficult but not impossible because they are in a ribbon cable) Also, the K40 mirror alignment is brutal to get right (crappy mirror mounts made out of bent aluminum) , so most K40's have a sweet spot of performance in the upper left corner.

funinthefalls, 2016-02-07:
I also will have more time this coming week to flesh out the documentation for you a little more, I've managed to stay out of the hospital for a few days now, so I think I will be around this week to work on it.

openhardwarecoza, 2016-02-07:
Then its still a firmware swop.

Configure the switch where it sits, as a MAX endstop in firmware, and set homing direction to Max 
Then set your software endstop for Min and axis lengh correctly (:  - and viola

openhardwarecoza, 2016-02-07:
Sorry to hear about the health problem! If theres anything I can do please shout!

funinthefalls, 2016-02-07:
You can get me a new spinal cord, heart, knees and some lungs.Just a few issues, but I will never let them stop me from enjoying my life. Anyways, everything is fantastic when you are on 520mg of morphine a day :) Do you still have the 40w tube and mirror donation? My tube seems to be OK, but I can't stand the mirror mounts that came with my K40.

openhardwarecoza, 2016-02-07:
:( sadly they havent made those open source yet 

Sorry to hear!  

We can buzz Dustin and see if he can help with that tube. Paypal has been dead quite this month but I have it penciled in to help!

cojarbi, 2016-02-07:
Admire your spirit man. Keep on living

openhardwarecoza, 2016-02-09:
The 'shortest path for the laser' argument is a valid reason though.... let me sleep on the idea, may actually make sense to think of top right as 0,0

funinthefalls, 2016-02-09:
Top left, not top right. :)

DamianTrejtowicz, 2016-03-24:
i sort it out this problem
all what you need to do is connect Y axis end stop ramps pin y max endstop,
then change in configuration.h:

// ENDSTOP SETTINGS:
// Sets direction of endstops when homing; 1=MAX, -1=MIN
#define X_HOME_DIR -1
#define Y_HOME_DIR 1 //TU ZMIENIONE
#define Z_HOME_DIR -1

#define min_software_endstops true // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops true  // If true, axis won't move to coordinates greater than the defined lengths below.
// Travel limits after homing

// China Town K40 CO2 Laser Engraver/Cutter
#define X_MAX_POS 330
#define X_MIN_POS 0
#define Y_MAX_POS 210
#define Y_MIN_POS 0
#define Z_MAX_POS 75
#define Z_MIN_POS 0

openhardwarecoza, 2016-03-25:
I've decided to can this one once and for all.  Your machine MUST be setup correctly:   X+ = right,   X- = left.   Y+ = top, Y- = bottom.   Why?  Because: https://en.wikipedia.org/wiki/Cartesian_coordinate_system

[image]

**This enhancement DOES NOT NOR EVER WILL reverse the directions of the cartesian standard!!!!**

What the ORIGIN refers to is do you want to position your material at 0,0, or 0,MAX = since laser beams loose power over distance there is merit to moving the origin to the top.  BUT that means the top corner will still be for example, Y 400 (in the case of a Freeburn2 laser).    But we'll work the moves in a reverse order, so counting down from Y-MAX instead of up from 0....   

So please, pay attention and stop arguing like idiots about where a machines MIN and MAX are (;   

- MINIMUM = LOWER LEFT
- MAXIMUM = UPPER RIGHT

Gustavvr, 2016-03-25:
@openhardwarecoza I completely agree with your decision to use the standard cartesian coordinate system. 

The issue is K40 are pretty bad when not in the upper left quadrant.  Would you be willing to entertain a transform option to that moves the image to MaxY MinX.  This is something I can work on.  But I don't want to work on it if you are going to deny the eventual pull request.

openhardwarecoza, 2016-03-25:
Thats exactly what we discussed above ^   -  make the origin configurable.
Translation is easy:

Draw a bounding box on the object.  And translate by laserymax - bbox.max.y
(:

Peter

openhardwarecoza, 2016-03-25:
And i appreciate help,  but consider laserweb1 end of life from development
perspective,  so please implement this in LaserWeb2 (:

openhardwarecoza, 2016-03-25:
See https://github.com/openhardwarecoza/LaserWeb2/issues/8 for the ongoing dev.  Closing this issue (:

closed 2016-03-25 by openhardwarecoza