LaserWeb/deprecated-LaserWeb1 #105: Add raster and vector cut the in same job

cojarbi, 2016-02-10:
At the moment this need to be done separately.  
If svg color detection is also integrated that would be awesome. 
Could be a good enhancement and makes the process easier.

openhardwarecoza, 2016-02-10:
We cannot raster SVGs (note how its absent from the Raster menu filetypes under the accept= near the end of line https://github.com/openhardwarecoza/LaserWeb/blob/master/i/index.html#L188)  and similarly we cannot vector cut from bitmaps.,. So you would still need to load two files. 

loading two files leads to offset andf positioning issues.

So handling it as seperate tasks would be the most reliable way of doing this since you can zero out between operations, check where its going to do what using the Bounding Box, etc

closed 2016-02-10 by openhardwarecoza

funinthefalls, 2016-02-10:
I create a master file in Inkscape that has both the raster and vector components, separate them into 2 different layers. Then I output 2 separate gcode file s, one for raster, one for vector.  Run the raster first , do not move the workpiece, then run the vector. Siunce both gcode files were created from the same master file, as long as you do not move the workpiece there will not be any alignment issues.

cojarbi, 2016-02-10:
Thanks @funinthefalls.  I this the way the industry handles this?

funinthefalls, 2016-02-10:
Not sure, that is just the way I do it.