LaserWeb/deprecated-LaserWeb1 #110: module.js:340     throw err;     ^

mkeyno, 2016-02-19:
hi guys 
I install according to  windows 7 64 instruction but when in final step type the node server , following error pop up;
any help is really appreciated 

Error: Cannot find module 'C:\Users\Private\LaserWeb\node_modules\serialport\bui
ld\Release\node-v47-win32-x64\serialport.node'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.Module._load (module.js:289:25)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (C:\Users\Private\LaserWeb\node_modules\serialport\ser
ialport.js:14:25)
    at Module._compile (module.js:425:26)
    at Object.Module._extensions..js (module.js:432:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)

openhardwarecoza, 2016-02-19:
> at Object. (C:\Users\Private\LaserWeb\node_modules\serialport\ser

Hey.. What flavour of Windows?

Just to recap on the steps

1.  You did remember the 'npm install'  step,  and was from within the
LaserWeb directory so that it could read package.json. Thats pretty much
the only thing you could be doing wrong.

2.  If you are a non-vanilla windows (i.e long since fresh install,  other
software that could interfere)  you may be unable to install the
precompiled node-serialport modules.   The project we use from upstream
node-serialport has instructions on compiling (using python and the windows
SDK) ,  here: https://github.com/voodootikigod/node-serialport/wiki that
you can try,  to install the dependency from source

Peter

mkeyno, 2016-02-19:
hi Peter
thanks for quick reply but I cant fully understand what is your mean , however I do "npm install serialport" and then "npm install" again but this time have less error 

`Error: Cannot find module 'C:\Users\Private\server'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.Module._load (module.js:289:25)
    at Function.Module.runMain (module.js:457:10)
    at startup (node.js:138:18)
    at node.js:974:3`

is there anythings that should be installed before staring the server?

by the way have you aware about cool guys that use ESP8266 to wireless communication ? I mean if the software can connect the wifi module which is connected to serial port of 3D printer , then there is no necessary to dig down to the serial port on windows

please visit these links

http://creatorbot.com/projects/electronics/add-esp8266-wifi-to-your-3d-printer

https://github.com/luc-github/ESP8266

openhardwarecoza, 2016-02-19:
Go check my googleplus profile ,  my wifi cnc/printer/laser prototypes came
in (:  you really should follow me there.

John lauer,  ray Kholodovsky and myself,  are just quickly in a conf call
working on chilipeppr stl/dxf import...  Else i woulve said lets jump in a
hangout.  But too busy at the moment.  Do you have spaces in your path?
Where did you install?

Try copy laserweb to c:\laserweb

Open a command prompt

Type cd c:\laserweb
Run npm install
Type dir and paste me a screenshot here

Run node server.js

mkeyno, 2016-02-19:
great Peter , the future own the only wireless links and i saw many guys try to use this chip wifi for many links to their machines , actually , I did build my first CNC laser with modified marlin + RAMPS1.4 which is connected with HC-06 BT, I successfully print  through Repetier-host , however after couple of weeks find great Igorr code with ESP module and currently use it for couple of my IOT device , then find your awesome web base platform for create the G-code and send it over the serial connection and I think the combination of your work and guys such as  Luc https://github.com/luc-github to create transparent wifi connection would the great ,

openhardwarecoza, 2016-02-19:
Woe thanks for sharing that firmware,  looks nicer than ESPlink.

mkeyno, 2016-02-19:
Thanks Peter , can address me your wifi cnc printer , I cant not find it in your wordpress or G+ page ? also have you try to add your tools in inkspace , ?

mkeyno, 2016-02-19:
[image]
[image]

cojarbi, 2016-02-19:
@mkeyno Try cloning again from github. I just did the whole procedure on a new machine without issues.

openhardwarecoza, 2016-02-19:
Follow me on google plus.   All my links are on about.me/peter6960  (but
google. Plus is my best)

mkeyno, 2016-02-19:
@cojarbi  not yet buddy , same error
[image]

openhardwarecoza, 2016-02-19:
Still doesnt look like you have node-serialport installed

1, you are sure you have node 0.12 not node 4 or 5?

and 

2, screenshot of npm install serialport

openhardwarecoza, 2016-02-19:
and wait by the look of that window, is that XP?

mkeyno, 2016-02-19:
I have node ver v5.6.0 Stable  and windows 7 x64 , should I remove the the prior version? how can I install this version

openhardwarecoza, 2016-02-19:
Yes, thats what the instructions say after all! (:
We wouldnt say 0.12 if serialport worked with 5! 

(: For not reading instructions feel free to see our https://github.com/openhardwarecoza/LaserWeb/wiki/SPONSORS page to ease the Guilt for this unneccesary Issue (;

openhardwarecoza, 2016-02-19:
For future ref, see https://github.com/voodootikigod/node-serialport/issues/578#issuecomment-150016435 

And from yours above, looks like they still havent fixed it for Windows (;

closed 2016-02-19 by openhardwarecoza

mkeyno, 2016-02-19:
sorry about that but how can I install this version of node , ? there is no good instruction for windows platform

mkeyno, 2016-02-19:
you know I heard about it  long time a go ,  how is hard to working with hardware resource in windows platform, I think you and your friends must  have new plan for using the wifi link instead of serial port

cojarbi, 2016-02-19:
@mkeyno the instructions are very clear. Click the link and Download node.v0.12.7-x86.msi

mkeyno, 2016-02-19:
tnx @cojarbi  but I have download  node-v0.12.7-darwin-x64.tar.gz which is no binary in it , ok I'll try this version, however as previous note I've just curious  to dig around your CAM environments and my ambitions is to use wifi link not serial port

openhardwarecoza, 2016-02-19:
See https://nodejs.org/en/download/releases/