LaserWeb/deprecated-LaserWeb1 #111: error: Invalid gcode ID:24 Aborted Job!  Safety First!

karel314, 2016-02-22:
First of all thx for this Great platform. I've been using GRBLweb to drive my homebuild GRBL(0.9i) based laser  engraver from a Rpi 2 for almost a year now. But seeing LaserWeb I could not wait to try it.

Installation on the Rpi2 was straightforward. Manual controls seem to work. However, if I sent a GCODE which has previously successfull been sent with GRBLWeb, the job aborts with the error:

error: Invalid gcode ID:24
Aborted Job!  Safety First!

Can you point me to the right direction to solve this? Thx.

cojarbi, 2016-02-22:
Karel

There are some issues with serial communication for GRBL at the moment. There were some random moves generated by this so this security was enabled. 

Ariel Yahni

karel314, 2016-02-22:
Ok, so I should just this issue: Serial Resend Debugging affecting Grbl 0.9J mostly #94
Should we close this issue? 

Does it help if I try to debug? Let me know? 

I can try to reproduce this issue tomorrow night (almost bedtime here).

cojarbi, 2016-02-22:
@karel314 look for https://github.com/openhardwarecoza/LaserWeb/issues/94

openhardwarecoza, 2016-02-23:
Also see, https://plus.google.com/+PetervanderWalt/posts/19Qz4QowESa   porting over.  Slowly.... See sponsors page on wiki if you are in a rush (:

closed 2016-02-23 by openhardwarecoza