LaserWeb/deprecated-LaserWeb1 #113: Raster strange engraving paths

nextime, 2016-02-23:
Hello,
i'm trying to rasterize some images, and then engraving them, but i get unexpected results.

On the 3d view it seems ok, but when i engrave it, there are some paths added.

Any idea on why does that? i'm not sure if it's a bug or something i'm doing wrong...

[image]

[image]

openhardwarecoza, 2016-02-23:
Which firmware are you using?  Looks like the grbl bug.   If its marlin,
print from SD

nextime, 2016-02-23:
It's marlin, i don't have any sd attached :|

nextime, 2016-02-23:
So, do you think it can be something related to how laserweb send gcode? 

Maybe i can try to send it by using pronterface...

nextime, 2016-02-23:
Confirmed, using pronterface to send gcode it works like a charm, so, it seems to be an issue with the serial implementation in laserweb/node.js?

openhardwarecoza, 2016-02-23:
Marlin cant keep its serial buffer full.. Upgrade to a smoothie or add an
SD Card

openhardwarecoza, 2016-02-23:
Also,  see the other open issues here

nextime, 2016-02-23:
I don't think it's a marlin issue as using another host ( pronterface ) the issue disappear, seems more between node.js a serial

openhardwarecoza, 2016-02-23:
Thats why i say,  look at the other open issues here...  Instead of opening
a duplicate to argue  (;

openhardwarecoza, 2016-02-23:
See

https://github.com/openhardwarecoza/LaserWeb/issues/94

https://github.com/voodootikigod/node-serialport/issues/663

The past week worth of work busy porting to chilipeppr
https://plus.google.com/+PetervanderWalt/posts/19Qz4QowESa

A whole community
https://www.google.co.za/url?sa=t&source=web&rct=j&url=https://plus.google.com/communities/118113483589382049502&ved=0ahUKEwjtgLGukI7LAhUC1hQKHRQGCnIQFggYMAA&usg=AFQjCNEZwbWNPiy0I7Jh525hnMYD81dxRA&sig2=sxjco0bEOSrX8rGyA5CJQQ

Yet,  you rather user your effort to complain than to contribute  (;

closed 2016-02-23 by openhardwarecoza

nextime, 2016-02-23:
sorry if it seems to be a complain, it isn't really, probably a language barrier ( i'm not native english ), but i was just trying to understand investigate to eventually, if i will be able to, even contribute with patches.

Thanks anyway for your pointers

openhardwarecoza, 2016-02-23:
(: cool.  Goodluck!