LaserWeb/deprecated-LaserWeb1 #115: Not connecting

ellisgl, 2016-03-26:
Info: Linux Lite (X64 - Ubuntu based), Node 4.4.1

Web shows that the device isn't connected, but it shows this when I start the server.
INFO:  Enabling webcam Widget
Connecting to /dev/ttyUSB0 at 115200
INFO:  Found device:  1a86 /dev/ttyUSB0 Firmware Detected: Grbl 0.9j   Port ID No:  32
Got OK
Queue Length: 0
Missing error handler on `socket`.
TypeError: Cannot read property 'port' of undefined
    at Socket.<anonymous> (/home/ellisgl/LaserWeb/server.js:811:26)
    at emitOne (events.js:77:13)
    at Socket.emit (events.js:169:7)
    at Socket.onevent (/home/ellisgl/LaserWeb/node_modules/socket.io/lib/socket.js:330:8)
    at Socket.onpacket (/home/ellisgl/LaserWeb/node_modules/socket.io/lib/socket.js:290:12)
    at Client.ondecoded (/home/ellisgl/LaserWeb/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/home/ellisgl/LaserWeb/node_modules/socket.io/node_modules/socket.io-parser/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/home/ellisgl/LaserWeb/node_modules/socket.io/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/home/ellisgl/LaserWeb/node_modules/socket.io/lib/client.js:175:18)
    at emitOne (events.js:77:13)
Got OK
Queue Length: 0
Got OK
Queue Length: 0

openhardwarecoza, 2016-03-26:
> at Socket. (/home/ellisgl/LaserWeb/server.js:811:26)

Look at the closed issues: Grbl 0.9 doesnt work so well :(

LaserWeb2 is coming soon.

openhardwarecoza, 2016-03-26:
https://github.com/openhardwarecoza/LaserWeb/issues/94 is what I am referring to

Try smoothie / marlin instead
(With Marlin, be sure to increase the serial buffer as per @nextime 's instructions: https://github.com/openhardwarecoza/LaserWeb/issues/94#issuecomment-188303108  )

In the meantime, sit tight, and keep a close eye on https://github.com/openhardwarecoza/LaserWeb2/commits/gh-pages (check every few days)  - in LaserWeb2 I am dumping node-serialport and switching to serialportjson server

closed 2016-03-26 by openhardwarecoza

openhardwarecoza, 2016-04-08:
Grbl is starting to come together:  See https://github.com/openhardwarecoza/LaserWeb2/issues/1