LaserWeb/deprecated-LaserWeb1 #116: Laser ON/OFF Widget

DamianTrejtowicz, 2016-03-26:
Is it possible to add laser on/off button to machine control widget?

openhardwarecoza, 2016-03-26:
Cant imagine why youd want to?

DamianTrejtowicz, 2016-03-26:
when you do mirror alligment i need give laser puls examply for 20ms

openhardwarecoza, 2016-03-26:
wire up a TL button like the rest of the world. Much safer than using software for laser alignment.

closed 2016-03-26 by openhardwarecoza

prcdslnc13, 2016-03-28:
Universal does their alignment pulse like this. With the soft button and
time. I'll be honest. Its sketchy as hell. I really like having a physical
button I can push.

funinthefalls, 2016-03-28:
Marlin uses either or. You can wire up a test button, like it originally had, and that will peacefully coexist with the software driven test feature. The nice thing about the soft button is you can set time and powerlevel, more versatile than a standard test button.

prcdslnc13, 2016-03-28:
I was thinking you could do a reallllllllly tiny move and combine the move
with a feed that would give you the burn time your after and the ability to
set power in software.

Not perfect but I think it will work. Gonna test tonight maybe.