LaserWeb/deprecated-LaserWeb1 #118: Add support for Drag knife

rafffel, 2016-04-07:
Can be cool to enhance you're open software to drag knife.
i have on the way an open source plotter with drag knife.
i actually use DXF2Gcode(https://sourceforge.net/projects/dxf2gcode/?source=navbar) for generate the drag knife gcode with swiwel specific due to offset rotating cutter
Source is avaible

openhardwarecoza, 2016-04-07:
Definitely will be adding!  (:

openhardwarecoza, 2016-04-07:
Offset is already available,  check github.com/openhardwarecoza/LaserWeb2

Enable the Tool Offset under the gear icon on LaserWeb2 -> Application
Settings.

Whats missing for proper drag knife is some simple Z moves.  I'll try
adding that tonight if time allows

rafffel, 2016-04-07:
Wow good to ear :)

the most important part in drag knife is swiwel rotation : http://www.cambam.co.uk/forum/index.php?action=dlattach;topic=3628.0;attach=11582

Due to offset of knife (that's allow to knife to rotate into the microbearing) for rotate the knife to cut path need some offset in gcode path specific to drag knife/
It's was really tricky move that i have didn't noticed until i have done my own plotter and try a simple gcode out of simple 3d printer slicer ^^'

i'm going to check the you're laserweb2 :)

openhardwarecoza, 2016-04-07:
Nice demonstration of what's needed -  though that's not in there yet.  But
do have a play with what is,  and lets see how far off we are (:

openhardwarecoza, 2016-04-07:
Looks like we have something to see how its calculated here
http://donektools.com/free-cnc-router-software/links-2/

rafffel, 2016-04-07:
https://sourceforge.net/p/dxf2gcode/sourcecode/ci/master/tree/source/core/stmove.py

the main calculation in DXf2Gcode is function make_swivelknife_move at line 145 
The logic is dependen of 2 parameter angle threshold for do a swiwel move and offset of knife (offset from center of microbearing)
you need to go a bit far from the rotation point once (endpoint + offset) then do a rotation around this point for rotate knife in the right angle to drag and drag to next threshold angle detected.

rafffel, 2016-04-07:
maybe need to move this enchancement to laserweb2 if i got some time next week i will look more deeper in you're code for add some code about plotter drag knife (like swiwel if it isn't done and tweak for pressure control).

FYI my plotter is at his state :  https://www.youtube.com/watch?v=r97FQ8E5q-I

But z motor doesn't great IMHO i would move to a servo motor with gear for control pressure applied by spring.
And i want to get an eady vinyl adhesive load (so that's my problem :p )

I also use Marlin firmware With ramps for control all of that i will look about gcode customisation (seem to view some possibility already done :) ).

openhardwarecoza, 2016-04-07:
Any coding help is welcomed with open arms!

openhardwarecoza, 2016-04-08:
Gave you push access to the LaserWeb2 repo

closed 2016-05-11 by openhardwarecoza