LaserWeb/deprecated-LaserWeb1 #120: Add support to MarlinKimbra firmware

GordTheRogue, 2016-04-13:
Hello, (first comment ever in GitHub) I'm tring to use LaserWeb on a cartesian XY machine which uses R.A.M.P.S. shield like a classic RepRap 3D printer.
I installed the MarlinKimbra firmware on the Arduino and I'm tring to figure how to get LaserWeb recognize it since is Marlin based. 
The Server says:
```
ERROR: Error connecting to webcam: connect ECONNREFUSED 127.0.0.1:8080 - Disabling Webcam Widget
INFO:  Found device:  Arduino LLC (www.arduino.cc) COM18 Firmware Detected: undefined   Port ID No:  0
Got OK
Queue Length: 0
ERROR:  Error from machine: Error:SD init fail
Aborted Job - safety first!
INFO: Machine Profiles read from file
Got OK
Queue Length: 0
INFO:  Now interacting with  COM18  running  undefined
Checking for Updates
Got OK
```
On the web page I get SERVER: OK - DEVICE NOT CONNECTED so I cannot interact with the machine or send anything to it.
How can I do? Or can you modify so MarlinKimbra could be recognized?
I'm suggesting to do this since MarlinKimbra is probably going to be the official release of Marlin for the 32bit boards.
More info at https://github.com/MagoKimbra/MarlinKimbra
Thanks a lot for everything

openhardwarecoza, 2016-04-13:
Kimbra is a 3D printing firmware.    Try out TurnkeyTyranny's Marlin for
Lasers.

Or even better,  switch to Smoothie (; Marlin sucks

openhardwarecoza, 2016-04-13:
See https://github.com/TurnkeyTyranny/buildlog-lasercutter-marlin

closed 2016-04-13 by openhardwarecoza