LaserWeb/deprecated-LaserWeb1 #123: Pinout of Stepper?

ajhalls, 2016-04-29:
Sorry for a stupid question, but I am playing around with this and got the SPJS installed, the webapp finds the port for the Arduino and I can connect to it without any issues. 

I currently just have a couple stepper drivers plugged in, no laser, just 2 SBT0811 controllers with 2 28BYJ-48 steppers. I have a smoothieboard at home somewhere (not sure ATM where) - I wanted to build a CNC mill or lathe someday but haddn't gotten around to it yet. I wanted to play around with the idea again and so starting with some basics seemed best.

I don't see anywhere to configure which which pins are the stepperX, which is stepperY, which is the relay for power and so on.

I have some Raspberry Pi's available as well, but thought I would just start with my windows PC so I didn't have to disconnect a monitor if possible - but if needed, it is available.

openhardwarecoza, 2016-04-29:
Firmware specific question.   Thats configured in firmware,  not in
LaserWeb

LaserWeb 1 (where you posted this issue)  does not use SPJS....

LaserWeb2 does but still very beta

cojarbi, 2016-04-29:
So we can assume you are running marlin ( hate it hate it) ? Not the best way to start even though is older. Find your Smoothieboard ( priceless ) you will get to the cool part of playing faster.

Once your there, come back for the rest.

ajhalls, 2016-04-29:
Sounds good, thank you for letting me know. I will take off work a little early today and see if I can find it. 

I am very sorry, but I am more of a software person than hardware, this is actually my first real effort at playing with the arduino. When you say that is a firmware question, I assume I need to configure the firmware for listening for specific commands, like a node API? If that is the case, do I need to match up my "variable" names with what you have them defined as?

cojarbi, 2016-04-29:
Firmware is where you will assign a logical pin to a physical one. With smoothieboard you just need to modify a txt config file and drop it on the SD. That's it, no compiling, etc.

ajhalls, 2016-05-02:
I checked all over this weekend and couldn't find my smoothieboard. I ordered a new one as well as a knockoff to play with: http://www.ebay.com/itm/221710223813?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

I will get back with you guys when the parts arrive.

closed 2016-05-11 by openhardwarecoza