LaserWeb/deprecated-LaserWeb1 #124: Part of firmware?

ajhalls, 2016-05-10:
Just a quick check. I got my smoothieboard and looked at their web interface, while I realize you could get Laserweb running on a RaspberryPi or other webserver, have you looked into just making it part of the firmware to replace the existing web interface?

https://github.com/Smoothieware/Smoothieware/tree/8cbd981e85c918e059a6e68d70fbf3cdad0f8ca5/src/libs/Network/uip/webserver

I myself am not a C+ developer, but I could ask around to see what would be needed to include a node engine along with, or instead of, the httpd one.

Just a thought :)

openhardwarecoza, 2016-05-10:
See github.com/arthurwolf/fabrica that i am a part of too  along with
github.com/openhardwarecoza/LaserWeb2  (no more Nodejs) - only a matter of
time

openhardwarecoza, 2016-05-10:
https://docs.google.com/document/d/1xuLbjVvCkizSkmPvfprBgnlKV7jBn7wQnLto_G5Q-Dw/edit?usp=sharing
= Fabrica Spec sheet:    note the paragraph about the Make screen (;

ajhalls, 2016-05-10:
Sounds awesome, thanks for all your hard work. I will keep thinking about how I can contribute.
Alan

closed 2016-05-11 by openhardwarecoza