LaserWeb/deprecated-LaserWeb1 #125: Add support for Laos devices

grbd, 2016-05-31:
At our hackspace we have a lasercutter setup with Laos
http://redmine.laoslaser.org/projects/laos/wiki/SimpleCode
I think it would be usefull if Laserweb had support for the laos protocol (to replace visicut)

openhardwarecoza, 2016-05-31:
We also have a Laos based machine at our Hackerspace.   Since a lot of the
code in the cam side generates compliant gcode,  and even worse the 3D
viewer parses gcode,  implementing simplecode will take a lot of work.  I
am happy to help you out where to go digging,  if you are willing to do the
digging,  testing,  etc to get it working before we take it to the
production version?

openhardwarecoza, 2016-05-31:
Ypu basically need to

1.    Add a Laos buffer to Serial-port-json-server
2.  Make the protocol drop down value an if statement in three2gcode and
laserraster.js so instead of gcode,  you create simplecode
3.  Add parsing to gcode-parser.js for the viewer.

openhardwarecoza, 2016-07-21:
Closing.  SimpleCode used by Laos is too far out.   Get a Smoothieboard (;

closed 2016-07-21 by openhardwarecoza